brettdottech/info-orbs #126: Certain stocks not showing correct currency

brett-dot-tech, 2024-11-26:
@racingsquirrel outlined that when they use the BARC ticker, it shows up as $ despite being pounds. I hacked together the currency matching logic and sounds like I just need to put in a few more redundancies for markets that dont pass it in the payload. I'll fix sometime this week (:

Omertron, 2025-01-24:
Certain stocks also seem to use mixed case, e.g. `https://api.twelvedata.com/quote?symbol=PAY&exchange=LSE`
has `"currency":"GBp"` 
A quick hack to the code fixes the correct symbol, but perhaps converting all to uppercase in the setCurrencySymbol function would fix this and any other errors?

HugoBalder, 2025-01-28:
I will open a Pull request for this fix, because it's really simple and only one line must be added in the StockDataModel.cpp:

StockDataModel &StockDataModel::setCurrencySymbol(String currencySymbol) {
    **currencySymbol.toUpperCase();**

Omertron, 2025-01-28:
> I will open a Pull request for this fix, because it's really simple and only one line must be added in the StockDataModel.cpp:
> 
> StockDataModel &StockDataModel::setCurrencySymbol(String currencySymbol) { **currencySymbol.toUpperCase();**

Thank you, I don't know C++ well enough to work out what needed to be done. I did find that location though :)

HugoBalder, 2025-02-02:
As you can see, the pull request is open. I hope that soon someone will care about bug fixing in this project.

flattermann, 2025-03-02:
Fixed by #235

closed 2025-03-02 by flattermann