brettdottech/info-orbs #148: Make network calls async

flattermann, 2024-12-01:
We should try to make the network calls asynchronous.

At the moment, the UI is completely blocked during updates. This is bad for UX, because the buttons don't work and the Orbs are "frozen".

lyricnz, 2024-12-01:
Also, it would be good to be able to switch from the "loading screens" to the first widget before all the widgets had updated their data (faster bootup time)

brett-dot-tech, 2024-12-10:
@dreed47 made some huge progress here, but keeping this issue open as the async fix has resulted in some minor UX bugs #171 , and we have yet to roll it out globally to other widgets (weather). @learn-more suggested some sort of globalized function to handle such async calls which I think sounds like a good idea

flattermann, 2024-12-12:
We just have to make sure that only one HTTP request is running at a time, because a HTTPClient with ArduinoJson parsing will easily use 70-100kB RAM and at the moment, the Orbs only have ~130kB free RAM when idle.
So if we switch all widgets to async/tasks, this will become an issue at boot.

dreed47, 2024-12-12:
Maybe we could use a Task semaphore to only allow one HTTP request at a time.  It would require us moving all the HTTP requests to use Tasks.  Or possible create a HTTP helper class or method the we use instead that wraps the Task and semaphore stuff and maybe JSON filtering too?

lyricnz, 2025-01-11:
This is essentially #222 now, right?

dreed47, 2025-01-11:
yes I think so

flattermann, 2025-03-02:
Implemented for WeatherWidget, StockWidget and ParqetWidget now.

closed 2025-03-02 by flattermann