brettdottech/info-orbs #156: Middle Second 2 dot flash not working

Dragonfire3296, 2024-12-04:
Hi All,

In my board the middle 2 dot flash for the seconds tick seems to have issues after some time, I will add a video to show.

https://github.com/user-attachments/assets/4fa3b3ed-ec49-4102-ad06-12611f606771

Dragonfire3296, 2025-01-14:
I have the latest DEV build running and just noticed that this has started again after about 5 days. Any troubleshooting I can help with?

HugoBalder, 2025-02-05:
Tried it again, startet the orb, everything was loaded fine and then i disabled my hotspot on my iphone (have to use this, as the wifi in the office has a firewall).

After it had no internet anymore, the colon in the middle is refreshed sporadically and the second dot jumps only every 2 seconds.

So finally everyone can reproduce it - no wifi, lagging middle screen.

Dragonfire3296, 2025-02-05:
I know on mine it is still connected to the network though...is there a wAy to double check?Get Outlook for AndroidFrom: HugoBalder ***@***.***>Sent: Wednesday, February 5, 2025 7:49:48 a.m.To: brettdottech/info-orbs ***@***.***>Cc: Dragonfire3296 ***@***.***>; Author ***@***.***>Subject: Re: [brettdottech/info-orbs] Middle Second 2 dot flash not working (Issue #156)
Tried it again, startet the orb, everything was loaded fine and then i disabled my hotspot on my iphone (have to use this, as the wifi in the office has a firewall).
After it had no internet anymore, the colon in the middle is refreshed sporadically and the second dot jumps only every 2 seconds.
So finally everyone can reproduce it - no wifi, lagging middle screen.

HugoBalder, 2025-02-05:
I would connect the esp32 to my pc and in visual code run the serial monitor and look on the output. Maybe in your environment you have at some time more than one device with the same IP address. 
Look if you see the lines i posted with beginpacket and endpacket. This stuff is blocking the orb. Don‘t know who implemented it, maybe this person should try, if this behaviour can be changed. Now as we know how to get to this error, it is simple to reproduce it.

HugoBalder, 2025-02-06:
Now i have found a second reason, why this occurs. 
It also happens, when the ntp-server pool.ntp.org cannot be reached. In the office our wifi with the firewall can load the stocks and the weather, but the ntp-port is blocked. Therefore also the second tick is updated only every 2 seconds and the colon is not redrawn.

Dragonfire3296, 2025-02-06:
> A blocking network call sounds like a very reasonable explanation why this sometimes happens, and only for some people.

I am checking now, I have the monitor running.

Dragonfire3296, 2025-02-10:
Finally got it to go, but different error, same vein:
Drawing of Clock took 455 ms
[97613397][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11
[190450963][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11
[225515407][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11
[246186984][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11
[254554744][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11
[327625046][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11

Dragonfire3296, 2025-02-10:
I wonder if we just right in a clause to restart if it detects errors?

HugoBalder, 2025-02-10:
Maybe you have a DNS Server set in your router like 8.8.8.8 instead of the dns server from your provider. In the info orb wifi settings we cannot set any dns server. Only a few people posted this error in the github repository of the wifimanager.

ScottWell1, 2025-02-24:
I occasionally see this same issue (colon stops flashing).

In my environment, all NTP calls are intercepted and handled by the router's internal ntpd daemon.  Loss of WAN entirely, or loss of DNS resolution to pool.ntp.org, or lack from response from pool.ntp.org, would have no effect here.  As long as the router is up, ntp requests are all handled locally.

Could a blocking network call (with failed DNS or no response) from another widget, like weather or stocks, be the cause?  I didn't think those requested any data until cycling to that widget, but I don't know the code that well.