brettdottech/info-orbs #198: Questions

Mikede78, 2024-12-26:
1. Can I edit the config to make changes after its setup? If so how? 
2. Is there an option to turn off the red light on the ESP32 chip? 
3. Are there any clock images besides the default one?

HugoBalder, 2024-12-27:
1. the config cannot be edited after it is flashed to the esp, but on the Webpage you can change tons of settings, when the device is on.
2. the red LED is directly connected to power and cannot be turned on/off by software. You have to desolder it.
3. Join the discord channel, there are many links and discussions about custom clock faces.

flattermann, 2024-12-31:
1. If you flash the newest dev firmware, you can open the WebPortal once your Orbs are connected to your Wifi: http://ip-of-your-orbs
2. As @HugoBalder said, that's not controllable by software
3. Joining the discord is always a good idea. :-) Here are some links to custom clocks:

https://github.com/AndreaFontolan/Info-Orbs-Clockfaces
https://info-orbs-clock-style-showcase.pages.dev/

In the newest dev version, you can go to the "File Browser", open a CustomClock directory and directly paste one of the URLs from the repos above to let the Orbs download the clock images, e.g.

https://github.com/AndreaFontolan/Info-Orbs-Clockfaces/tree/main/Binary%20clock%201
or
https://info-orbs-clock-style-showcase.pages.dev/images/10/

closed 2024-12-31 by flattermann