brettdottech/info-orbs #199: "Error; please specify "upload_port"

azey23, 2024-12-27:
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1

Anyone seen this error before? Cannot find where to specify

azey23, 2024-12-27:
Thanks Hugo - I still cannot seem to locate this in platformo.ini 

Is there a screenshot anywhere I can use? Thanks

HugoBalder, 2024-12-27:
You have ro add a new line with upload_port = COM<number> in this section of platformio.ini
[image]

azey23, 2024-12-27:
Next problem; I don't have usb's numbered in device manager. (COM3, ect) 

I have "Universal Series bus controllers" but they name driver types. 

Is there an alternative solution?

//acer Nitro 5

HugoBalder, 2024-12-27:
Watch the Youtube Video from Brett and start at Minute 13, there is everything described how to flash and install everything.

azey23, 2024-12-27:
Yes, didn't help unfortunately. However, I connected USB-C to USB-C and it worked! In case anyone else struggling with this

closed 2024-12-27 by azey23