brettdottech/info-orbs #223: Inaccurate timing for first stock rendering

lyricnz, 2025-01-12:
No, it didn't take 93 seconds to render the BTC/USD ticker - it was on the previous screen

```
I: Left button short pressed -> switch to prev Widget
displayStock - BTC/USD ~ 93879.93
displayStock - USD/CAD ~ 1.44
displayStock - XEQT ~ 33.56
displayStock - SPY ~ 580.49
displayStock - APC&country=Germany ~ 228.55
Drawing of Stock took 1221 ms
```

lyricnz, 2025-01-12:
Cycling around all the pages of widgets, and back to the ticker, displays _precisely_ the same values. Suspicious?

```
I: Left button short pressed -> switch to prev Widget
displayStock - BTC/USD ~ 93879.93
displayStock - USD/CAD ~ 1.44
displayStock - XEQT ~ 33.56
displayStock - SPY ~ 580.49
displayStock - APC&country=Germany ~ 228.55
Drawing of Stock took 1221 ms
I: Left button short pressed -> switch to prev Widget
Drawing of Weather took 755 ms
I: Left button short pressed -> switch to prev Widget
Drawing of Clock took 459 ms
I: Left button short pressed -> switch to prev Widget
displayStock - BTC/USD ~ 93879.93
displayStock - USD/CAD ~ 1.44
displayStock - XEQT ~ 33.56
displayStock - SPY ~ 580.49
displayStock - APC&country=Germany ~ 228.55
Drawing of Stock took 1221 ms
```

flattermann, 2025-01-12:
The numbers are not timings, they are the stock prices, e.g. 93879.93 $ per BTC. :-)

lyricnz, 2025-01-13:
Hahahah, ok!

closed 2025-01-13 by lyricnz