brettdottech/info-orbs #224: Output of "garbage" on serial port

lyricnz, 2025-01-12:
The lines of silly character may be mangled by my terminal, but something is being output there:

```
I: ✅ Released semaphore
I: 🔵 Starting HTTP request for: https://api.twelvedata.com/quote?apikey=e03fc53524454ab8b65d91b23c669cc5&symbol=APC&country=Germany
I: ✅ Released semaphore
����������������������������������������������������������������1
I: Left button short pressed -> switch to prev Widget
displayStock - BTC/USD ~ 93879.93
displayStock - USD/CAD ~ 1.44
displayStock - XEQT ~ 33.56
displayStock - SPY ~ 580.49
displayStock - APC&country=Germany ~ 228.55
Drawing of Stock took 1221 ms
����������������������������������������������������������������
```

dreed47, 2025-02-28:
I don't see these on my terminal.  Is anyone else seeing them?  I'm on MacOS

dreed47, 2025-03-01:
I have a theory that these garbage characters are a result of electrical interference.  Let me explain why I say that.  First, I hardly ever see them and I understand many people see them all the time.  This leads me to believe its a local environmental issue.  Second they seem to appear at random points in the app execution.  Third they seem to print without a line feed.  I did a search through the code to find places where we do a Serial.print (not Serial.println) and there are only a few and none of them seem like likely candidates.   Electrical interference can come from bad usb cables, nearby devices, motors, or power supplies. Long cables act as antennas, picking up noise and degrading the signal.  Adjacent wires interfering with each other can corrupt the data.

flattermann, 2025-03-02:
Startup garbage character should be taken care by #246.

But we cannot really do something about electrical interference. :-)

Feel free to reopen if necessary.

closed 2025-03-02 by flattermann

lyricnz, 2025-03-04:
FYI: che and I spent quite a bit of time yesterday digging into this - observing the patterns, cutting out code and features (all the way to turning off wifi, all the widgets, and even emptying the main loop()), and still having the same issue.  Finally, we connected the device directly to PC, and not via USB hub, and problem went away.