brettdottech/info-orbs #225: Frequent errors about timezone updates, including errors

lyricnz, 2025-01-12:
I happen to be on the stock ticker screen, with default config.h

```
Next timezone update: 1741514613
Timezone Offset from API: -28800
Next timezone update: 1741514536
Timezone Offset from API: -28800
Next timezone update: 1741514642
Timezone Offset from API: -28800
Next timezone update: 1741514596
Timezone Offset from API: -28800
Next timezone update: 1741514554
Timezone Offset from API: -28800
Next timezone update: 1741514560
Timezone Offset from API: -28800
Next timezone update: 1741514578
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Timezone Offset from API: -28800
Next timezone update: 1741514607
Timezone Offset from API: -28800
Next timezone update: 1741514537
Timezone Offset from API: -28800
Next timezone update: 1741514637
Timezone Offset from API: -28800
Next timezone update: 1741514695
Timezone Offset from API: -28800
Next timezone update: 1741514584
Timezone Offset from API: -28800
Next timezone update: 1741514505
Timezone Offset from API: -28800
Next timezone update: 1741514444
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Deserialization error on timezone offset API response
Timezone Offset from API: -28800
```

flattermann, 2025-01-12:
That's strange. It should only update the TZ Offset twice per year (after DST change).

```
if (m_timeZoneOffset == -1 || (m_nextTimeZoneUpdate > 0 && m_unixEpoch > m_nextTimeZoneUpdate)) {
  getTimeZoneOffsetFromAPI();
}
```

Clearly, the `m_timeZoneOffset` was fetched/set correctly (to -28800) and the `m_nextTimeZoneUpdate` is also correct (1741514xxx).

lyricnz, 2025-01-13:
```
Next timezone update: 1741514613
Timezone Offset from API: -28800
Next timezone update: 1741514536
Timezone Offset from API: -28800
Next timezone update: 1741514642
Timezone Offset from API: -28800
Next timezone update: 1741514596
Timezone Offset from API: -28800
```

FWIW:
- 1741514613 = Sun Mar 09 2025 10:03:33 GMT+0000
- 1741514536 = Sun Mar 09 2025 10:02:16 GMT+0000
- 1741514642 = Sun Mar 09 2025 10:04:02 GMT+0000
- 1741514596 = Sun Mar 09 2025 10:03:16 GMT+0000

So the next-update is jumping around a bit, and is still 2 months in the future.  What's causing it to get reset so often, or queried so often?

lyricnz, 2025-01-13:
I added some debug:

```cpp
void GlobalTime::updateTime(bool force) {
    if (force || millis() - m_updateTimer > m_oneSecond) {
        Serial.println("GlobalTime::updateTime - inside1");
        if (m_timeZoneOffset == -1 || (m_nextTimeZoneUpdate > 0 && m_unixEpoch > m_nextTimeZoneUpdate)) {
            Serial.printf("GlobalTime::updateTime - inside2: m_timeZoneOffset: %d; m_nextTimeZoneUpdate=%d; m_unixEpoch=%d\n", m_timeZoneOffset, m_nextTimeZoneUpdate, m_unixEpoch );
            getTimeZoneOffsetFromAPI();
        }
```

And it went from just doing inside1, to doing the actual update (for no obvious reason):

```
GlobalTime::updateTime - inside1

GlobalTime::updateTime - inside1

GlobalTime::updateTime - inside1

GlobalTime::updateTime - inside1

GlobalTime::updateTime - inside2: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514533; m_unixEpoch=2085949696
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514420
GlobalTime::updateTime - inside1

GlobalTime::updateTime - inside2: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514420; m_unixEpoch=2085949697
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514573
GlobalTime::updateTime - inside1

GlobalTime::updateTime - inside2: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514573; m_unixEpoch=2085949698
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514668
GlobalTime::updateTime - inside1
```

That m_unixEpoch looks wrong:
```
1741514407 Sun Mar 09 2025 10:00:07 GMT+0000
1741514651 Sun Mar 09 2025 10:04:11 GMT+0000
2085949740 Wed Feb 06 2036 22:29:00 GMT+0000
```

lyricnz, 2025-01-13:
Change the debug code to:
```cpp
        Serial.printf("GlobalTime::updateTime - inside1: m_timeZoneOffset: %d; m_nextTimeZoneUpdate=%d; m_unixEpoch=%d\n", m_timeZoneOffset, m_nextTimeZoneUpdate, m_unixEpoch );
        if (m_timeZoneOffset == -1 || (m_nextTimeZoneUpdate > 0 && m_unixEpoch > m_nextTimeZoneUpdate)) {
            Serial.println("GlobalTime::updateTime - inside2");
            getTimeZoneOffsetFromAPI();
        }
```

And it starts out ok:
```
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514587; m_unixEpoch=1736721898
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514587; m_unixEpoch=1736721900
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514587; m_unixEpoch=1736721901
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514587; m_unixEpoch=1736721902
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514587; m_unixEpoch=1736721903
```

Waiting for it to go bad now....

lyricnz, 2025-01-13:
Still haven't hit it yet. "Something" is definitely corrupting m_unixEpoch

lyricnz, 2025-01-13:
I just rebooted the orbi, which triggered the same issue (probably a problem calling web API)

```
WifiManager connected.[  5167][E][Preferences.cpp:483] getString(): nvs_get_str len fail: timezoneLoc NOT_FOUND
[  5185][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ntpServer NOT_FOUND
GlobalTime initialized, tzLoc=America/Vancouver, clockFormat=1, ntpServer=pool.ntp.org
[  5198][E][Preferences.cpp:483] getString(): nvs_get_str len fail: weatherLocation NOT_FOUND
WeatherWidget initialized, loc=Victoria, BC, mode=1
[  5211][E][Preferences.cpp:483] getString(): nvs_get_str len fail: stockList NOT_FOUND
Constructing ParqetWidget, portfolioId=
[  5225][E][Preferences.cpp:483] getString(): nvs_get_str len fail: pqportfoId NOT_FOUND
New config section: General
New config section: TFT Settings
New config section: ClockWidget
New config section: WeatherWidget
New config section: StockWidget
New config section: ParqetWidget
IP address: 192.168.101.71
*wm:Starting Web Portal 

Connected to WiFi
updating widget Clock
updating widget Weather
updating widget Stock
GlobalTime::updateTime - inside1: m_timeZoneOffset: -1; m_nextTimeZoneUpdate=0; m_unixEpoch=0
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514594
I: 🔵 Starting HTTP request for: https://weather.visualcrossing.com/VisualCrossingWebServices/rest/services/timeline/Victoria, BC/next3days?key=REMOVED&unitGroup=metric&include=days,current&iconSet=icons1&lang=en
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514594; m_unixEpoch=2085949696
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514637
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514637; m_unixEpoch=2085949697
```

then it came right after a while (with some of that garbage output)

```
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514520
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514520; m_unixEpoch=2085949818
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514468
����������������������������������������������������������������: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514468; m_unixEpoch=2085949820
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514580
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514580; m_unixEpoch=2085949823
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514489
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514489; m_unixEpoch=2085949825
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514446
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514446; m_unixEpoch=1736723119
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514446; m_unixEpoch=1736723120
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514446; m_unixEpoch=1736723121
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514446; m_unixEpoch=1736723122
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514446; m_unixEpoch=1736723123
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514446; m_unixEpoch=1736723124

```

lyricnz, 2025-01-13:
Feels like a bug in error handling in NTPClient. Maybe https://github.com/arduino-libraries/NTPClient/issues/105#issuecomment-656154072

lyricnz, 2025-01-13:
NTP sync returns random day and time for the year 2036
https://github.com/gmag11/NtpClient/issues/70

> Research shows that the time/date 07:28:16 07/02/2036 is the NTP roll-over date. The time stamp in NTP is given in seconds, starting from 1.1.1900 (UNIX starts 1.1.1970).

Which is what we're seeing.

lyricnz, 2025-01-13:
> I think I am starting to understand. There is a method called getTime. This method requests a NTP message from an NTP server and returns the new time. But if there is a timeout, it raises a noResponse ntpEvent and then it returns 0; This 0 then gets interpreted as 02/07/2036.

lyricnz, 2025-01-13:
Yeah it just happened again:

```
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514687; m_unixEpoch=1736775418
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514687; m_unixEpoch=1736775419
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514687; m_unixEpoch=2085949696
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514508
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514508; m_unixEpoch=2085949697
GlobalTime::updateTime - inside2
Fetching Timezone Offset from API: 
Timezone Offset from API: -28800
Next timezone update: 1741514587
```

lyricnz, 2025-01-13:
It looks like we're hitting the NTP server every minute - that's way too often. 

```cpp
        bool wasUpdated = m_timeClient->update();
        m_unixEpoch = m_timeClient->getEpochTime();
        if (wasUpdated) {
            Serial.printf("GlobalTime::updateTime - inside3 - got result %d\n", m_unixEpoch);
        }
```

emits

```
GlobalTime::updateTime - inside3 - got result 1736779208 # Mon Jan 13 2025 14:40:08 GMT+0000
GlobalTime::updateTime - inside3 - got result 1736779268 # Mon Jan 13 2025 14:41:08 GMT+0000
GlobalTime::updateTime - inside3 - got result 1736779329 # Mon Jan 13 2025 14:42:09 GMT+0000
```

lyricnz, 2025-01-13:
Showing error happening from subsystem:

```cpp
    if (force || millis() - m_updateTimer > m_oneSecond) {
        Serial.printf("GlobalTime::updateTime - inside1: m_timeZoneOffset: %d; m_nextTimeZoneUpdate=%d; m_unixEpoch=%d\n", m_timeZoneOffset, m_nextTimeZoneUpdate, m_unixEpoch );
        if (m_timeZoneOffset == -1 || (m_nextTimeZoneUpdate > 0 && m_unixEpoch > m_nextTimeZoneUpdate)) {
            Serial.println("GlobalTime::updateTime - inside2");
            getTimeZoneOffsetFromAPI();
        }
        bool wasUpdated = m_timeClient->update();
        m_unixEpoch = m_timeClient->getEpochTime();
        if (wasUpdated) {
            Serial.printf("GlobalTime::updateTime - inside3 - got result %d\n", m_unixEpoch);
        }
```

emitting

```
lobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514694; m_unixEpoch=1736782302
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514694; m_unixEpoch=1736782304
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514694; m_unixEpoch=1736782306
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514694; m_unixEpoch=1736782308
GlobalTime::updateTime - inside3 - got result 2085949696
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514694; m_unixEpoch=2085949696
GlobalTime::updateTime - inside2
Timezone Offset from API: -28800
Next timezone update: 1741514399
```

The 208... is "Wed Feb 06 2036 22:28:16 GMT+0000"

We could probably just do a check for date >2030 and ignore it (or retry).

flattermann, 2025-01-14:
This looks like a simple fix, but we would need to use our own patched NTPClient:

https://github.com/arduino-libraries/NTPClient/pull/211/files

flattermann, 2025-03-02:
Fixed by #245

closed 2025-03-02 by flattermann