brettdottech/info-orbs #226: Don't query public NTP server every 60 seconds - this is too often

lyricnz, 2025-01-13:
It looks like we're hitting the NTP server every minute - that's way too often (antisocial).

```cpp
        bool wasUpdated = m_timeClient->update();
        m_unixEpoch = m_timeClient->getEpochTime();
        if (wasUpdated) {
            Serial.printf("GlobalTime::updateTime - inside3 - got result %d\n", m_unixEpoch);
        }
```

emits

```
GlobalTime::updateTime - inside3 - got result 1736779208 # Mon Jan 13 2025 14:40:08 GMT+0000
GlobalTime::updateTime - inside3 - got result 1736779268 # Mon Jan 13 2025 14:41:08 GMT+0000
GlobalTime::updateTime - inside3 - got result 1736779329 # Mon Jan 13 2025 14:42:09 GMT+0000
```

This is the default interval in NTPClient(), and there is a (separate) issue with getting back invalid times from that - see #225

flattermann, 2025-01-14:
Maybe change it to once per hour?

`void NTPClient::setUpdateInterval(unsigned long updateInterval)`

lyricnz, 2025-01-14:
Yes, but we also need to check if we have a valid time, and poll more often until that's done (there's first-time setup, and errors when NTP returns 0).

closed 2025-03-02 by flattermann