brettdottech/info-orbs #227: Invalid time on startup until NTP query succeeds

lyricnz, 2025-01-13:
On startup, the m_unixEpoch is set to some strange value (-28788 here) which persists until the clock updates via NTP some time later.  If we don't have a real (reliable) value, we should use a zero/undefined, and not use it to display widgets etc.  Related to #226 and #225 

```
Connected to WiFi
updating widget Clock
updating widget Weather
updating widget Stock
GlobalTime::updateTime - inside1: m_timeZoneOffset: -1; m_nextTimeZoneUpdate=0; m_unixEpoch=0
GlobalTime::updateTime - inside2
Timezone Offset from API: -28800
Next timezone update: 1741514528
I: 🔵 Starting HTTP request for: https://weather.visualcrossing.com/VisualCrossingWebServices/rest/services/timeline/Victoria, BC/next3days?key=XW2RDGD6XK432AF25BNK2A3C7&unitGroup=metric&include=days,current&iconSet=icons1&lang=en
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514528; m_unixEpoch=-28788
GlobalTime::updateTime - inside2
Timezone Offset from API: -28800
Next timezone update: 1741514656
I: ✅ Released semaphore
...
Next timezone update: 1741514540
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514540; m_unixEpoch=-28703
GlobalTime::updateTime - inside2
Timezone Offset from API: -28800
Next timezone update: 1741514563
GlobalTime::updateTime - inside3 - got result 1736780184
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514563; m_unixEpoch=1736780184
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514563; m_unixEpoch=1736780185
GlobalTime::updateTime - inside1: m_timeZoneOffset: -28800; m_nextTimeZoneUpdate=1741514563; m_unixEpoch=1736780186
```

flattermann, 2025-03-03:
#245 ignores invalid dates (before 2025 and after 2035).

lyricnz, 2025-03-04:
Will test this again tomorrow. We got distracted chasing the "noise" on the serial terminal

lyricnz, 2025-03-04:
I think #262 is working well, will do a little more testing, then maybe move the logging-time to be the NTP time, not the local/hardware offset since boot.

lyricnz, 2025-03-05:
Closing issue - #264 proposes using local-time, but the fundamental request is resolved (when `LOG_TIMESTAMP` is defined)