brettdottech/info-orbs #233: PCB Dimensions and layout

johnwargo, 2025-01-30:
Can you share the pcb measurements file?  I want to 3d print a mounting bracket for the board

learn-more, 2025-01-31:
The complete pcb is shared, that should contain the info you need?
[[info-orbs/tree/main/PCBFiles%2BSchematics/Main%20PCB](url)](https://github.com/brettdottech/info-orbs/tree/main/PCBFiles%2BSchematics/Main%20PCB)

johnwargo, 2025-01-31:
No, it does not, that's why I asked.  I'm not a PCB designer, so I can't pull the information I need out of the files.  Many PCB file sets include an image file that lays out the dimensions, that is what I'm asking for. Like this

[image]

HugoBalder, 2025-01-31:
The pcb is 194x28mm, just measured it 😀

johnwargo, 2025-01-31:
Yes, I have the dimensions of the board, what I want is just like I showed in the sample image I provided - all of the measurements for the board and holes so I can build an enclosure for it. That means hole dimensions and placement as well. 

I assumed that could just be exported from the project, I just don't know anything about board design tools to do that.

HugoBalder, 2025-02-01:
Then buy simply a kit and you have everything 🤦

johnwargo, 2025-02-01:
@HugoBalder that is not helpful.  I have the board already.

Is it that bad that I’m asking for the exact measurements? If you don’t want to help me, then fine, but I thought this was a community that I could ask for help in.

johnwargo, 2025-02-01:
Let me start over. Hi, I'm @johnwargo. I'm new to this project but I'm a long time [Arduino](https://johnwargo.com/categories/arduino/), Raspberry Pi, [ESP32](https://johnwargo.com/categories/esp32/) developer and a contributor to multiple projects on GitHub. 

I purchased the Info Orbs kit and I absolutely love it.  I want to make an enclosure for it but I'm not sure yet whether I'm going to 3D print one or build it out of wood, perhaps cut with a LASER cutter. I have no experience with PCB design tools and very little experience with 3D design tools.

I'm looking for the complete dimensions for the board.  I was able to extract just the board layer from the PCB files, so I have the outside dimensions I need. What I'm looking for is the specific location and diameter of the bolt holes on the board as well as the positions and dimensions of the external components (power connector, buttons, female header pins for the displays). With all of that information, I can easily design a 3D printable or LASER cuttable enclosure (and, of course, share it with the community).  

Yes, I know that I can measure them on the board, but I'd prefer to not do that if I can.  I also know that some PCB design tools allow you to export/save an image of the board with the dimensions laid out like the image I shared earlier in this Issue. If this is possible, can someone with the experience with PCB design tools is on this discussion either extract those dimensions and share them here so others can leverage them as well or export a file like the one I showed above and add it to the repository so it's part of the project going forward?

If you can't help me with this request, I understand. If you can, I'd be grateful. Failing that, I'll just measure it myself and deal with the potential inaccuracies that come with manual measurement.

lyricnz, 2025-02-19:
Hi @johnwargo . Is there some problem with just installing KiCad and then opening the design files, then using the measurement tool to show you dimensions you're interested in? Eg

[image]

I am also not a PCB designer - I literally installed the software to answer this question.

lyricnz, 2025-02-19:
@johnwargo Also you can use the Dimension tools to add multiple measurements on your PCB

[image]

lyricnz, 2025-02-19:
https://docs.kicad.org/6.0/ru/pcbnew/pcbnew_inspecting.html#measurement_tool

https://docs.kicad.org/6.0/ru/pcbnew/pcbnew_editing.html#dimensions

[image]

johnwargo, 2025-02-19:
@lyricnz no, there isn't a problem with me installing KiCad myself and figuring it out, I already mentioned that in my rewritten response. I don't want to learn KiCad, but I know that I could. 

You and @HugoBalder (in a deleted response) are so quick to chastise me about not doing it myself that you're missing the bigger opportunity here. When you look online, you'll find that many Info Orbs owners designed and published 3D plans for enclosures for the device. I imagine that there are many others like me that built or are planning to build enclosures out of wood or perhaps other materials for the device. 

Because you seem so opposed to simply adding the dimensions to the open source repository, each and every one of these enthusiasts must perform those steps individually. Either manually measuring the board and hole placement or opening KiCad and generating the dimensions themselves. Think of how helpful it would be for the **Community** if it simply included that file as part of the regular distribution.

Rather than doing all that work to:

1. Download and install KiCad
2. Learn how to document the dimensions of the board
3. Write three chastising responses on my Github issue

You instead simply did some of that work (ignoring the chastising) then submitted the dimensioned image in a pull request on the repo. That would be less work for you and makes it easier for anyone who follows me wanting to ask the same question to get what they need without having to be chastised. 

I am so very sorry that I even asked my question. I'm used to working in open source repositories where the community **wants** to help rather than feel superior to new people on the forum.

lyricnz, 2025-02-19:
I don't know who you think "community" is, if it's not just people like you and I. Every single person here (except Brett) are just people who bought the kit, or liked the idea, and got involved. Some of us help, contribute, share. There are no workers to do this work - only us.  Your idea is useful - great - so do it yourself and contribute the result. That's open-source, brother.

johnwargo, 2025-02-19:
My sincerest and humblest apologies for even suggesting a change to the project.

lyricnz, 2025-02-22:
I attempted to add dimensions to the file, but I think I'm hitting some grid/snap issue, so some of the measurements look a bit off.  But it looks like the design isn't "perfect"?  Eg: look at the bottom of the purple boxes: one is above the line, one is on the line, one is below the line:

[image]

Which means that the corner-to-corner (4.6267) lengths aren't quite the same as the edge-to-edge (4.6250)?

[image]