brettdottech/info-orbs #234: Alternative power orientation

johnwargo, 2025-01-30:
I'm planning on building a wooden enclosure for my device and, for a variety of reasons, I don't want the power connector sticking out the back of the top of the enclosure. If the headers weren't installed on the USB connector, I could use an angled headers instead and point the power connector down into the enclosure, hiding it from the top back of the enclosure. 

So, that's my feature request.

From an implementation standpoint, can you tell me where you got the power connector board?  I want to see if I can get one without headers to solve my problem.

HugoBalder, 2025-01-31:
I have mine from Amazon -> https://www.amazon.de/dp/B0C6D3MXK9

johnwargo, 2025-01-31:
yeah, that works. Here it is for US: https://www.amazon.com/XUGERIP-Breakout-Type-C-Connector-Converter/dp/B0DFQD7XWH