brettdottech/info-orbs #252: Add (optional?) timestamp to log output

lyricnz, 2025-03-03:
This would help us identify the timing when working on issues that may be related (eg: network or NTP or background updates etc)

lyricnz, 2025-03-03:
Right now, output is a mixture of `Log` output and `Serial.print`.  Making output a bit of a mixed bag:

[image]

lyricnz, 2025-03-03:
Q: should the logging use the native time on the device (which appears to start at 00:00:00 - see above), or the time we get from NTP (GlobalTime)?

flattermann, 2025-03-04:
Implemented in #261

closed 2025-03-04 by flattermann

lyricnz, 2025-03-04:
I'm going to add support for printing the NTP time, not just the hardware/offset time.