brettdottech/info-orbs #282: Duplicate stock symbols in StockWidget shows "loading data"

lyricnz, 2025-03-15:
If I want to have the same symbol shown more than once, that should work ;)  But the second request for data shows
`E: Duplicate Task. Task already in the queue to waiting to be processed.`
And the widget isn't drawn properly.

This becomes a slightly-less silly edge-case when using >1 page of stocks per #268

dreed47, 2025-03-15:
Yeah the key for tasks is the url. We could could probably just append something to that like the screen number.  But also from an efficiency standpoint is there any reason to go fetch data you are already requesting?  I.e should the stock widget be changed to either not allow dup's or somehow share data between them?

flattermann, 2025-03-15:
I think it would make sense to use the same data for both screens, without requesting the data twice.