brettdottech/info-orbs #283: Make ClockWidget widget

lyricnz, 2025-03-15:
No reason why it should be hardcoded. Maybe I don't want a clock?

dreed47, 2025-03-15:
I Agree. I think all widgets should behave the same including the ability to turn them off

flattermann, 2025-03-15:
We can do that, we just need to make sure that something like `WidgetSet::next()` does not go into an infinite recursion.
At the moment, it expects at least one widget to be enabled.

https://github.com/brettdottech/info-orbs/blob/f584d09f14ae5186ff52b0548ba1efbe62c22908/firmware/src/core/widget/WidgetSet.cpp#L43

dreed47, 2025-03-15:
WidgetSet::next() should first check the count
```cpp    
if (m_widgetCount == 0) {
        Log.warningln("No widgets available");
        return;
    }
```

flattermann, 2025-03-16:
That only helps if you compile with 0 widgets (and that will probably "never" happen, but we should check for it anyway).

But we also need to check for the case that a user **disables** all widgets in the WebUI. That might actually happen by accident.

nconner-06, 2025-03-17:
don't forget WidgetSet::prev() :-(

maybe better to check m_enabled as this is what the user can change..  Once we add the ability to change it's value in the UI.