brettdottech/info-orbs #296: incorrect time shown on clock when switching widgets

nconner-06, 2025-03-16:
when you change widgets, the clock widget briefly shows the old time before refreshing

This is caused because WidgetSet::switchWidget, is calling "setup" rather than "update":

void WidgetSet::switchWidget() {
    m_screenManager->clearAllScreens();
    getCurrent()->**setup**();
    uint32_t start = millis();
    getCurrent()->draw(true);
    uint32_t end = millis();
    Serial.printf("Drawing of %s took %d ms\n", getCurrent()->getName().c_str(), (end - start));
}

should be getCurrent()->update();

flattermann, 2025-03-17:
Please do not remove `setup()` here, because this is also used by the widgets to know that they are about to be displayed.
e.g. the ParqetWidget uses it to always start with the first screen.

(And yes, we should implement a better lifecycle management, e.g. like android (onCreate(), onResume(), onPause()...)

But adding `.update()` after `.setup()` here should work.

nconner-06, 2025-03-17:
ok, but for me, setup should only be called ONCE when the orb starts and not every time we cycle to it..

flattermann, 2025-03-17:
Yes, that's definitely more logical.

As soon as we implement some other lifecycle hooks (e.g. onResume()/onPause() if we cycle to/from a widget) we can remove it here.

dreed47, 2025-03-28:
calling getCurrent()->update(); at switchWidget is causing the clockwidget to draw with missing digits.  See issue #310 .  This change got moved into dev with PR #271    I submitted PR #312 to revert this in dev