brettdottech/info-orbs #298: use Lat/Long for Visual Crossing weather

nconner-06, 2025-03-16:
Standardize to Lat/Long values (where possible) for all weather feeds.

Currently Visual Crossing is using place name

lyricnz, 2025-03-17:
I think forcing end-users to enter lat/long is deeply un-usefriendly.  If the underlying APIs use it, we should consider using a lookup service to resolve the location name into lat/long.

dreed47, 2025-03-17:
> I think forcing end-users to enter lat/long is deeply un-usefriendly. If the underlying APIs use it, we should consider using a lookup service to resolve the location name into lat/long.

Visual Crossing already allows you to specify a location name.  Maybe provide Lat/Lon as an alternative for people that want to use it?