brettdottech/info-orbs #304: 5Zone widget and Async HTTP requests

dreed47, 2025-03-18:
While testing the 5zone widget, startup seems to take a lot longer due to waiting for initial time zone API requests.   Those API requests also seem to be locking up the UI in some cases.  These requests should be done async using TaskManager so startup is not delayed and UI doeesnt get locked up.  Also should include a default Loading message while the API's are returning.

HugoBalder, 2025-03-19:
Oh boy, the code quality and testing before even thinking to create a Pull Request is getting lower and lower when i once created my PR‘s :( Maybe it will get better in the future.

brett-dot-tech, 2025-03-19:
Thanks for this @dreed47, my bad for not catching this while merging. I'll work to see if I can fix on my end.

@HugoBalder while I appreciate the sentiment of your comment it's not helpful at all to anyone.
This is an open source project with folks of varying skill levels (of which I am on the lower end of) that I am doing my best to accommodate and make sure all willing to contribute are able to do so, and I'm still learning!

Suggestions, feedback, changes and help are VERY welcomed(+encouraged!!) and I do my best to accommodate any and all of them, however empty criticism is not helpful.

dreed47, 2025-03-19:
I didn't realize there was already a loading messages. I just didn't see them.  I get your point about having data preloaded but as a general rule any long running operation like an http request should be done async otherwise it completely blocks the main loop and everything appears locked up until it finishes. That's compounded when we have to do multiple http requests one for each screen

dreed47, 2025-03-19:
If we really wanted to home up the startup until all data is ready we should use a semaphore

flattermann, 2025-03-19:
I would prefer the startup to **not** be blocked, but to load all data in a background thread.

Especially if we add more widgets, the time-to-fully-load-all-widgets will go up, but IMO it's better if the user can at least see the clock while the data is still loading.

nconner-06, 2025-03-19:
I'm re-writing the startup to use the task manager, so as not to block the Orb

Also, building in the new timer function for the draw frequency

nconner-06, 2025-03-20:
PR #305 created to cover these points