brettdottech/info-orbs #311: Failed to connect to ESP32

jaybak, 2025-03-28:
After following the instructions, I can't seem to get the ESP32 to enter the write mode.  During the power up, the "boot" button is held and upload is started.  But I always get an error.

_
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM5
Uploading .pio\build\esp32doit-devkit-v1\firmware.bin
esptool.py v4.5.1
Serial port COM5
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https ://docs.espressif.com/projects/esptool/en/latest/troubleshooting. html
*** [upload] Error 2
============================ [FAILED] Took 21.84 seconds =====================
_

I can connect to the ESP32 via serial port COM5.  I've added both the universal and the basic windows drivers.  This error happens when the chip is connected to the circuit and when its isolated.  USB C cable is tested and is a good data cable.  VSC on an updated Win 11 PC.

How can I force the "write"  mode and upload the code?

flattermann, 2025-03-31:
Did you keep pressing the boot button from power up until the flashing process begins? (i.e. until Connecting..... )

jaybak, 2025-03-31:
> Did you keep pressing the boot button from power up until the flashing process begins? (i.e. until Connecting..... )

Yes with no luck.  Is it a bad chip or could some odd Win 11 setting blocking me?  The VSC can see the chip on COM5.