brettdottech/info-orbs #323: Compilation failed

kerat71, 2025-04-27:
Hi everyone,
I got stuck with new version of Info Orbs.
Here how PlatformIO stops:
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (54.3.20) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a 
 - tool-esptoolpy @ 4.8.9 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 3.2.0
|-- FS @ 3.2.0
|-- SPIFFS @ 3.2.0
|-- LittleFS @ 3.2.0
|-- SD @ 3.2.0
|-- NTPClient @ 3.2.1
|-- ArduinoJson @ 7.4.1
|-- StreamUtils @ 1.9.0
|-- TFT_eSPI @ 2.5.43
|-- TJpg_Decoder @ 1.1.0
|-- PubSubClient @ 2.8.0
|-- Time @ 1.6.1
|-- WiFiManager @ 2.0.17
|-- HTTPClient @ 3.2.0
|-- WiFi @ 3.2.0
|-- OpenFontRender
Building in release mode
Converting .pio/build/esp32doit-devkit-v1/logo.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/moonCloudW.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/sunCloudsW.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/sunW.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/moonW.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/snowW.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/rainW.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/cloudsW.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/moonCloudB.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/sunCloudsB.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/sunB.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/moonB.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/snowB.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/rainB.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/cloudsB.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/0.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/1.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/2.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/3.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/4.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/5.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/6.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/7.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/8.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/9.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/colon_on.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/colon_off.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_0.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_1.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_2.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_3.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_4.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_5.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_6.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_7.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_8.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_9.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_colon_on.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/custom_colon_off.jpg.txt.o
Converting .pio/build/esp32doit-devkit-v1/RobotoRegular.ttf.txt.o
Converting .pio/build/esp32doit-devkit-v1/FinalFrontier.ttf.txt.o
Converting .pio/build/esp32doit-devkit-v1/DSEG7ModernBold.ttf.txt.o
Converting .pio/build/esp32doit-devkit-v1/DSEG14ModernBold.ttf.txt.o
Compiling .pio/build/esp32doit-devkit-v1/src/core/button/Button.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/core/globaltime/GlobalTime.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/core/screenmanager/ScreenManager.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/core/utils/Utils.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/core/widget/ScreenWidget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/core/widget/Widget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/core/widget/WidgetSet.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/main.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/clockwidget/ClockWidget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/mqttwidget/MQTTWidget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/parqetwidget/ParqetDataModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/parqetwidget/ParqetHoldingDataModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/parqetwidget/ParqetWidget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/stockwidget/StockDataModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/stockwidget/StockWidget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/weatherwidget/WeatherDataModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/weatherwidget/WeatherWidget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElement.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementArcModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementCharacterModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementCircleModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementImageModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementLineModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementRectangleModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementTextModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataElementTriangleModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataModel.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/webdatawidget/WebDataWidget.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/src/widgets/wifiwidget/WifiWidget.cpp.o
Building .pio/build/esp32doit-devkit-v1/bootloader.bin
esptool.py v4.8.9
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generating partitions .pio/build/esp32doit-devkit-v1/partitions.bin
Compiling .pio/build/esp32doit-devkit-v1/lib4b1/SPI/SPI.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib2da/FS/FS.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib2da/FS/vfs_api.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib3bf/SPIFFS/SPIFFS.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib9e7/LittleFS/LittleFS.cpp.o
In file included from .pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.h:94,
                 from firmware/src/widgets/wifiwidget/WifiWidget.cpp:4:
/home/milen/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'?
  222 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                ^~
      |                                fs::FS
compilation terminated due to -Wfatal-errors.
Compiling .pio/build/esp32doit-devkit-v1/lib1d1/SD/SD.cpp.o
*** [.pio/build/esp32doit-devkit-v1/src/widgets/wifiwidget/WifiWidget.cpp.o] Error 1
Archiving .pio/build/esp32doit-devkit-v1/lib4b1/libSPI.a
Indexing .pio/build/esp32doit-devkit-v1/lib4b1/libSPI.a
============================================================= [FAILED] Took 8.41 seconds =============================================================

Please, help! What am I missing?

dreed47, 2025-04-28:
is this the main branch or the dev branch?  I would recommend you try the dev branch.

kerat71, 2025-04-28:
Yes, this is the main branch. I will try the dev branch to what will happen. Thanks for the answer.

HugoBalder, 2025-04-29:
I just downloaded the Main Version here - openend the Workspace in Visual Studio Code, waited until the spinning circle in the lower right corner stopped, copied the config.h.template to config.h and let it compile. Everything worked fine without errors.

My version of Visual Studio Code and Platform IO on a Windows 11 machine here:

Version: 1.99.3 (user setup)
Commit: 17baf841131aa23349f217ca7c570c76ee87b957
Date: 2025-04-15T23:18:46.076Z
Electron: 34.3.2
ElectronBuildId: 11161073
Chromium: 132.0.6834.210
Node.js: 20.18.3
V8: 13.2.152.41-electron.0
OS: Windows_NT x64 10.0.26100

Platform IO Version Core 6.1.18 - Home 3.4.4

So the Main branch is ok and can be compiled without errors.

kerat71, 2025-05-03:
Unfortunately, this doesn't work for me. Freshly setup VSCode with PlatformIO and copy of info-orb-main.
Nothing touched except config.h and this is what I get:
`Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (54.3.20) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a 
 - tool-esptoolpy @ 4.8.9
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 3.2.0
|-- FS @ 3.2.0
|-- SPIFFS @ 3.2.0
|-- LittleFS @ 3.2.0
|-- SD @ 3.2.0
|-- NTPClient @ 3.2.1
|-- ArduinoJson @ 7.4.1
|-- StreamUtils @ 1.9.0
|-- TFT_eSPI @ 2.5.43
|-- TJpg_Decoder @ 1.1.0
|-- PubSubClient @ 2.8.0
|-- Time @ 1.6.1
|-- WiFiManager @ 2.0.17
|-- HTTPClient @ 3.2.0
|-- WiFi @ 3.2.0
|-- OpenFontRender
Building in release mode
Converting .pio\build\esp32doit-devkit-v1\logo.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\moonCloudW.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\sunCloudsW.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\sunW.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\moonW.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\snowW.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\rainW.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\cloudsW.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\moonCloudB.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\sunCloudsB.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\sunB.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\moonB.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\snowB.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\rainB.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\cloudsB.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\0.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\1.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\2.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\3.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\4.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\5.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\6.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\7.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\8.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\9.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\colon_on.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\custom_0.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\custom_1.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\custom_2.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\custom_3.jpg.txt.o
Converting .pio\build\esp32doit-devkit-v1\custom_4.jpg.txt.o
[... 40 lines cut ...]
compilation terminated due to -Wfatal-errors.
*** [.pio\build\esp32doit-devkit-v1\src\widgets\clockwidget\ClockWidget.cpp.o] Error 1
In file included from firmware/src/main.cpp:2:
firmware/src/core/globaltime/GlobalTime.h:76:5: error: 'WiFiUDP' does not name a type
   76 |     WiFiUDP m_udp;
      |     ^~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
In file included from firmware/src/widgets/parqetwidget/ParqetWidget.h:4,
                 from firmware/src/widgets/parqetwidget/ParqetWidget.cpp:1:
firmware/src/core/globaltime/GlobalTime.h:76:5: error: 'WiFiUDP' does not name a type
   76 |     WiFiUDP m_udp;
      |     ^~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\esp32doit-devkit-v1\src\widgets\parqetwidget\ParqetWidget.cpp.o] Error 1
In file included from firmware/src/widgets/weatherwidget/WeatherWidget.h:4,
                 from firmware/src/widgets/weatherwidget/WeatherWidget.cpp:13:
firmware/src/core/globaltime/GlobalTime.h:76:5: error: 'WiFiUDP' does not name a type
   76 |     WiFiUDP m_udp;
      |     ^~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\esp32doit-devkit-v1\src\widgets\weatherwidget\WeatherWidget.cpp.o] Error 1
================================================================================ [FAILED] Took 12.54 seconds ================================================================================

 *  The terminal process "C:\Users\user\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. `

kerat71, 2025-05-03:
If I add "#include <WiFiUdp.h" in GlobalTime.h the compilation process ends up exactly like my OP:

`In file included from .pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.h:94,
                 from firmware/src/widgets/wifiwidget/WifiWidget.cpp:4:
C:/Users/user/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'?
  222 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                ^~
      |                                fs::FS
compilation terminated due to -Wfatal-errors.`

witch is most likely library issue?